arijav/AriloSprinkler-Firmware #21: use ESP32 local PIN for the 8 channel

chamroeunbm, 2021-04-25:
Hi,
ESp32 has enough pin for the 8 channels. Can you share me how to do that?

arijav, 2021-04-25:
Hi, in my board there are not enough pings for 8 channels due to the interface pins to the Lora E22 module. You can however create a board based on my design, removing the Lora module, the port expander and driving the triacs directly with the ESP32. If you create such a board have a look to the SW from J.Charer (https://github.com/JaCharer/OpenSprinkler-Firmware-ESP32port) as he is driving the outputs with ESP32 pins and no port expander

chamroeunbm, 2021-04-25:
Yes, i saw his code but it looks like there's no support for MQTT in his code, which is important. Yes, i plan to add NRF24. Are you sure that not enough PIN for 8 channels?

arijav, 2021-04-25:
In my board there is no place for sure. Below you can find a link to the table where you can see the use of each ESP32-WROOM pin:

https://docs.google.com/spreadsheets/d/1rRuH_Wl1ZeBDJMySVSdHYPLP0izNxznpmXPF5bGpdCk/edit#gid=0

chamroeunbm, 2021-04-26:
i update the front end app code because it is hardcoded in fhe main.js and the GPIO pin is working but not sure if this is the proper way.

closed 2021-11-15 by garudaonekh